OpenXiangShan/XiangShan #4630: expand load and store instruction

mlabaf2, 2025-04-25:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi
I expanded regfile length to 128bit to be able to read and write to regfile in 128bit whenever it is needed. Now I want to define CLD and CST to be able to load and store 128bit from/to memory. I want to implement them by calling two 64bit load and store instruction in the code. I added LDC and STC to DecodeUnit.scala. Where is the proper place to implement them? How can I implement these two instructions?

[image]

NewPaulWalker, 2025-04-28:
> I want to implement them by calling two 64bit load and store instruction in the code. I added CLD and CST to DecodeUnit.scala. Where is the proper place to implement them? How can I implement these two instructions?

This sounds very similar to the process of splitting complex instructions into micro-operations (uops). Your CLD and CST would be split into two uops, each responsible for a 64-bit load or store. I think you can refer to [DecodeUnitComp.scala](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/backend/decode/DecodeUnitComp.scala), which splits complex instructions into smaller-granularity uops.

mlabaf2, 2025-04-28:
> This sounds very similar to the process of splitting complex instructions into micro-operations (uops). Your CLD and CST would be split into two uops, each responsible for a 64-bit load or store. I [...]

Thank you for the response.
I am working on this branch "https://github.com/OpenXiangShan/XiangShan/tree/f580a020c2c5a94799a1df45c3e120f37acc2f83"
there is not DecodeUnitComp.scala module in code. Again, How can I implement CLD and CST?

NewPaulWalker, 2025-04-28:
Please confirm that you know this content. 
https://github.com/OpenXiangShan/XiangShan/issues/4556#issuecomment-2801041173

closed 2025-09-10 by Yan-Muzi