OpenXiangShan/XiangShan #4651: Metals unsupported Scala 2.13.15

ez4yunfeng2, 2025-04-30:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

When I use Vscode with Metals, Metals displays "WARN unsupported Scala 2.13.15". Can the official repository upgrade the Scala version to at least version 2.13.16 (and chisel-plugin needs to be upgraded to 6.7.0)?

Ma-YX, 2025-04-30:
Thanks for your issue.

As for Scala/Chisel version upgrade, we plan to carry out it in second half of this year.

As for the warning of Metals, I suggest you double check your settings, since Metals should support Scala 2.13.15 according to https://scalameta.org/metals/docs/editors/vscode/

```
Metals supports these Scala versions:

Right( - Scala 2.11: 2.11.12

Scala 2.12: 2.12.17, 2.12.18, 2.12.19, 2.12.20

Scala 2.13: 2.13.13, 2.13.14, 2.13.15, 2.13.16

Scala 3 versions from 3.3.4 are automatically supported by Metals.
```

ez4yunfeng2, 2025-04-30:
Thanks for the answer, when I switched Metals to a version prior to 1.48.0, this issue no longer appeared.

closed 2025-04-30 by ez4yunfeng2