OpenXiangShan/XiangShan #4655: Baremetal Programming, how to stop simulation

bantierr, 2025-05-01:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi,

I have looked around quite a bit, but I could not find any proper documentation on how to stop simulation in bare metal programs.

I've managed to get simulations to stop using what seems to be some form of custom instruction, `0x0005006b` by looking at the XiangShan project riscv-test repository. 
This successfully stops simulation, but I could not find what differentiate a `GOOD TRAP` from a `BAD TRAP`. It seems like my programs always stop when the instruction above (`0x0005006b`) is reached, but the trap is sometimes good sometimes bad. How can I make it so that it is a good trap.

Thank you for your time !
Best,
Quentin

Tang-Haojin, 2025-05-01:
The simulation hits a good trap when `rs0` register of the `xstrap` instruction holds the value of `0`, and a bad trap when it holds `1`. In this case of `0x5006b`, the `rs0` is `a0`. So when `$a0 == 0`, it hits a good trap.

bantierr, 2025-05-01:
Thank you very much for the very quick answer.

It works !

closed 2025-05-01 by bantierr

Tang-Haojin, 2025-05-01:
You are welcome. Happy International Workers' Day!