OpenXiangShan/XiangShan #4656: Dual Core Coherency Problem

bantierr, 2025-05-01:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

Hi, 

Sorry for the fast new issue. I am writing some bare-metal programs, and there is a scenario I do not get, which does not match with the spike behavior. 
I labeled this as a problem, because NEMU does not trigger an problem.

I have a simple program where:
1. 2 cores boot
2. Core 1 writes to a shared memory address and goes into low power mode (wfi)
3. Core 0 polls an address to see if Core 1 is in low power mode

But, in the commit logs, I can see that Core 1 stores to the correct address, but Core 0 never actually sees the value at that address and the programs stalls. 

### What did you do before

I double checked that both the rocket core and spike could run the program, which work with no issue. I do not think I have any coherency issue, as it is the only write to that address. 

I made a small repository: https://github.com/bantierr/xiangshan_issue to share the source code.

### Environment

- XiangShan branch: master
- XiangShan commit id: 16ae9ddcd
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Anzooooo, 2025-05-02:
Thanks for issues that this might be a XiangShan bug.
Can you provide your test source code and compiled in/elf file for us to further locate the problem.

bantierr, 2025-05-02:
Thank you very much for the fast answer !

Of course, the source code is in the repository I linked above (https://github.com/bantierr/xiangshan_issue), so feel free to compile it from source, there is a makefile. 
Just in case, here is the source code and a pre-compiled ELF file (I added the .txt extention so github did not complain):

[elf_program.txt](https://github.com/user-attachments/files/20009741/elf_program.txt)
[main_asm.txt](https://github.com/user-attachments/files/20009742/main_asm.txt)

I you do confirm it is a bug, I will naturally try to help.

Best,
Quentin

Anzooooo, 2025-05-02:
> Thank you very much for the fast answer !
>
> Of course, the source code is in the repository I linked above (https://github.com/bantierr/xiangshan_issue), so feel free to compile it from source, there [...]

Thanks, I'll confirm as soon as I can.

Anzooooo, 2025-05-14:
I'm sorry to get back to you so late, development work has been heavy lately.

There are two points to your question:

1. For modern processors (including XiangShan), the store instruction is not immediately written to cache/memory. Instead, it may exist in a buffer. For XiangShan, this module is called a `sbuffer`.
The `sbuffer` is not a coherent node, so the data that exists in the `sbuffer` is not observed by the other cores and the cache.
According to your test code, I found that you wrote data using hart 1, but then hart 1 went to wfi.
Therefore, according to XiangShan's design, this store will be written from sbuffer to cache after 1 million cycles.
Therefore, it takes 1 million cycles for hart 0 to observe hart 1's store.

2.
We provide for the termination of the simulation via the `xsTrap` instruction, i.e. `.insn 4, 0x0005006b`.
This will determine if it is `BAD TRAP` or `GOOD TRAP` based on the value of the a0 register.
If it is 0, then `GOOD TRAP`, but in your code, the value of a0 is the address that was previously written, and I don't see any other operations you are doing with a0.

---

To sum up, I suggest you to write the corresponding `fence` instruction after hart 1, which is also required by the manual, and only after `fence` you can guarantee multicore visibility. And, I suggest you manually set the finger of a0 manually before the `xsTRAP` instruction, to 0 if it exits normally, or non-0 (or maybe 1) if it doesn't.

bantierr, 2025-05-14:
Thank you very much for the answer ! 

I just tried it an it worked, I indeed did not wait for 1 million cycle when I tested, but the fence indeed flushed the store buffer properly.

closed 2025-05-14 by bantierr