OpenXiangShan/XiangShan #4667: Mismatch in `mcause` for Unaligned `amomin.d` Access Between XiangShan and Spike

LuLuji04, 2025-05-07:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。
- [x] I have reproduced the incorrect behaviors using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了不正确的行为。

### Describe the bug

When executing an unaligned `amomin.d` (atomic doubleword minimum) instruction, XiangShan and Spike report different `mcause` values. The address accessed is intentionally unaligned. However, the two simulators disagree on which exception should take priority:

- **Spike** reports: `mcause = 7` → **Store/AMO access fault**
- **XiangShan** reports: `mcause = 6` → **Store/AMO address misaligned**

[image]
[image]

### Expected behavior

As Spike reflects the correct trap priority: **access fault > misaligned address**. XiangShan should report `mcause = 7` (**Store/AMO access fault**) instead of `6`.

### To Reproduce

Here is the source code and binary file.
[test.zip](https://github.com/user-attachments/files/20091536/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 7e564dbbfb630d3142a5c023ab16922e0497be9d
- SPIKE : ready-to-run/riscv64-spike-so

NewPaulWalker, 2025-05-08:
XiangShan does not support unaligned atomic memory accesses, whereas in Spike, unaligned memory access is controlled by a global option.

https://github.com/OpenXiangShan/XiangShan/issues/3878

LuLuji04, 2025-05-08:
Thanks for the clarification! Much appreciated!

closed 2025-09-05 by Yan-Muzi