OpenXiangShan/XiangShan #4682: Fails to raise Instruction Access Fault on invalid PC

LuLuji04, 2025-05-11:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。
- [x] I have reproduced the incorrect behaviors using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了不正确的行为。

### Describe the bug

When executing a `jalr` instruction to an invalid PC (0x12345678), XiangShan continues to fetch and execute instructions, instead of raising an instruction fetch exception. In contrast, NEMU correctly identifies and reports the exception.
[image]
[image]

### Expected behavior

- XiangShan should raise an **Instruction Access Fault** (`mcause = 1`) upon fetching from an invalid PC.
- The `mtval` register should contain the faulting PC value (`0x12345678`).
- Execution should not continue from an invalid PC.

### To Reproduce

Here is the testcase.
[test.zip](https://github.com/user-attachments/files/20148359/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 0c97e1df8323fdab545e7d336867b444745299bd
- NEMU commit id: 16e9c675a07886f46cbbd48cf69e2e13eb919f9f

NewPaulWalker, 2025-05-12:
I think https://github.com/OpenXiangShan/XiangShan/issues/4386 can help you understand this inconsistency.

NewPaulWalker, 2025-05-12:
No! — please read this explanation more carefully https://github.com/OpenXiangShan/XiangShan/issues/4386#issuecomment-2710166242. XiangShan and NEMU handle flash addresses inconsistently. Also, you can try fetching and executing instructions from address 0x10000000 to see whether NEMU has flash.

> Ah, I understand. There is no flash simulation in NEMU, so it trigger `Instruction Access Fault`. But in Xiangshan it is legal to fetch instruction from flash.

LuLuji04, 2025-05-13:
I see, NEMU indeed includes flash support, which is where the first instruction is stored. However, why doesn't NEMU simulate a flash region of the same size as XiangShan (0x10000000 to 0x20000000)? Even if the actual flash hardware isn't that large, it shouldn't be difficult to simulate the entire address range as executable.

NewPaulWalker, 2025-05-15:
> I see, NEMU indeed includes flash support, which is where the first instruction is stored. However, why doesn't NEMU simulate a flash region of the same size as XiangShan (0x10000000 to 0x20000000)? [...]

https://github.com/OpenXiangShan/XiangShan/issues/4386#issuecomment-2712353089