OpenXiangShan/XiangShan #4689: XiangShan didn’t exit simulation in my test case

LuLuji04, 2025-05-14:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。
- [x] I have reproduced the incorrect behaviors using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了不正确的行为。

### Describe the bug

In the test case, after executing `mret`, the code jumps into the `_halt` function(provided by AM). However, the instruction `0005006b` did not cause XiangShan to exit to the simulation environment as expected. Instead, the instruction was executed, and even the following instruction was committed, until the simulation stopped due to a mismatch.

By contrast, NEMU exits successfully at the same point, despite hitting a bad trap:

[image]

XiangShan appears to ignore `0005006b`:

[image]

The simulation continues and only stops when a mismatch is finally triggered:

[image]

### Expected behavior

Upon executing the instruction `0005006b` inside the `_halt` function, the processor should immediately trigger a simulation exit.

### To Reproduce

Here is the testcase.
[test.zip](https://github.com/user-attachments/files/20201360/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 0c97e1df8323fdab545e7d336867b444745299bd
- NEMU commit id: 16e9c675a07886f46cbbd48cf69e2e13eb919f9f

NewPaulWalker, 2025-05-15:
I believe this issue is caused by a problem with the `TrapCode` in Difftest. https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/common/dut.h#L43

At that point, the value of `a0` is -1, instead of 0 or 1. In Difftest, -1 happens to be used to indicate "continue execution," so XiangShan proceeds with the simulation and executes the next instruction, `auipc`. Since Difftest determines that the simulation should continue, it also lets NEMU execute this instruction. However, NEMU directly terminates the program instead. As a result, you can observe an inconsistency: in NEMU (the "right" side), `a0` remains -1, while in XiangShan, it has been updated to the value after executing `auipc`.

LuLuji04, 2025-05-15:
Thanks for your confirmation!
Do you think it is necessary for NEMU to continue running after reporting an error code of -1, instead of terminating immediately?

NewPaulWalker, 2025-05-15:
> Thanks for your confirmation! Do you think it is necessary for NEMU to continue running after reporting an error code of -1, instead of terminating immediately?

Why not? Because we treat -1 as a signal to continue execution. If you want to halt the system, you can simply use 0 or 1.

LuLuji04, 2025-05-15:
It seems that this behavior is not currently implemented in NEMU. I suggest adding this feature to improve compatibility and consistency.