OpenXiangShan/XiangShan #4738: clock frequency

Liujiaqi-jlu, 2025-05-26:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello, I would like to ask what is the clock frequency of Nanhu v2.2 version,thanks!

eastonman, 2025-05-26:
Max clock frequency varies from 1GHz to 3GHz depending on process node and operating voltage.

Liujiaqi-jlu, 2025-05-27:
> Performance modeling work and the published materials are based on 2.0GHz for NanHu.

I am using the mirror version of Nanhu v2.2 and simulating it through emu. Is this simulation clock frequency also 2GHz?

rolypolytoy, 2025-07-25:
Simulation clock frequency via emu is often a lot slower than actual predicted hardware clock frequency, since emulation is a lot less efficient than the actual hardware will be.