OpenXiangShan/XiangShan #4778: Regarding the Measurement of DDR Interface Execution Rate

Liujiaqi-jlu, 2025-06-04:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello,I am using the Nanhu v2.2docker version of Xiangshan. I would like to use the workload to measure the speed of the DDR interface. The configuration program I wrote is as follows:memory copy is the measurement program, and the two address writing operations before and after are to observe the start signal and end signal for accurate time measurement. But it seems that there is no memory copy program execution time between the start signal and the end signal that I observed in the waveform diagram. The waveform diagram is shown below.I have tested these two instructions before, and when there are no other programs in between, the execution interval is also 38ps What is the reason for this?

[image]

[image]

[image]

Tang-Haojin, 2025-06-04:
> > Please look into your disasm for more details.
> 
> Sorry, what do you mean by disasm?

先看一下你编的程序的反汇编

Liujiaqi-jlu, 2025-06-05:
> > > Please look into your disasm for more details.
> > 
> > 
> > Sorry, what do you mean by disasm?
> 
> 先看一下你编的程序的反汇编

Thanks，i find the problem.