OpenXiangShan/XiangShan #4813: make verilog error

cc0h, 2025-06-15:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

When compiling the Verilog code, I keep encountering this error：

```
[814/814] =============== xiangshan.test.runMain t...fftest --full-stacktrace =============== 107s
==================================================================================================
1 tasks failed
xiangshan.resources os.SubprocessException: Result of git…: 1

    os.proc.call(ProcessOps.scala:230)
    build_.package_$xiangshan$.$anonfun$resources$3(build.sc:351)
make: *** [build/rtl/SimTop.sv] Error 1
```

Here is the complete log from my last build:

```
[uno@myServer XiangShan]$ make init
git submodule update --init
cd rocket-chip && git submodule update --init cde hardfloat
cd openLLC && git submodule update --init openNCB
[uno@myServer XiangShan]$ make sim-verilog
mkdir -p build/rtl
Mon, 16 Jun 2025 01:41:11 +0800
time -avp -o ./build/time.log mill -i -Djvm-xmx=40G -Djvm-xss=256m xiangshan.test.runMain top.SimTop    \
	--target-dir build/rtl --config DefaultConfig --issue E.b --firtool-opt "--repl-seq-mem --repl-seq-mem-file=SimTop.sv.conf"		\
	--num-cores 1 --dump-fir --target systemverilog --split-verilog --firtool-opt "-O=release --disable-annotation-unknown --lowering-options=explicitBitcast,disallowLocalVariables,disallowPortDeclSharing,locationInfoStyle=none"  --enable-difftest --full-stacktrace
[740/814] xiangshan.resources
[740] You need to run this command from the toplevel of the working tree.
[814/814] =============== xiangshan.test.runMain top....-difftest --full-stacktrace =============== 3s
1 tasks failed
xiangshan.resources os.SubprocessException: Result of git…: 1

    os.proc.call(ProcessOps.scala:230)
    build_.package_$xiangshan$.$anonfun$resources$3(build.sc:351)
make: *** [build/rtl/SimTop.sv] Error 1
```

### What did you do before

Due to poor network conditions, I executed `git clone --recursive https://github.com/OpenXiangShan/xs-env.git` and `source update-submodule.sh` on an Azure cloud server. Afterward, I compressed the entire` /xs-env `directory, transferred it to my own server, and extracted it there.

Since I'm using CentOS 7.9, I modified` setup-tools.sh` to compile `shadowsocks-libev` from source and installed other available dependency packages. Since I plan to use VCS for simulation, I did not install Verilator and the RISC-V toolchain, which should not affect Verilog compilation.

After extraction, I ran `source env.sh` in `/xs-env`, then executed `make init` and `make sim-verilog` in `/xs-env/XiangShan`.

### Environment

- XiangShan branch: master
- XiangShan commit id: c8bbd5c6acf879d0df9dcc3974562b54187254db
- XiangShan config: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: CentOS Linux release 7.9.2009 (Core)
- gcc version: gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
- mill version: Mill Build Tool version 0.12.3
- java version: Java version: openjdk 11.0.23 2024-04-16 LTS

### Additional context

I also tried `make help`, `make version`, and `make jar`, but encountered this error every time.

```
[uno@myServer XiangShan]$ make help
mill -i xiangshan.runMain top.TopMain --help
[673/723] xiangshan.resources
[673] You need to run this command from the toplevel of the working tree.
[671/723] rocketchip.compile
[671] [info] compiling 316 Scala sources to /home/xs/xs-env/XiangShan/out/rocketchip/compile.dest/classes ...
[671] [warn] 8 deprecations
[671] [warn] 67 deprecations (since 2.13.0)
[671] [warn] 1 deprecation (since 2.13.7)
[671] [warn] 1 deprecation (since 6.6.0)
[671] [warn] 1 deprecation (since 9)
[671] [warn] 52 deprecations (since CDE 0.1)
[671] [warn] 1 deprecation (since rocket-chip 1.2)
[671] [warn] 131 deprecations in total; re-run with -deprecation for details
[671] [warn] 19 feature warnings; re-run with -feature for details
[671] [warn] 9 warnings found
[671] [info] done compiling
[723/723] ========================= xiangshan.runMain top.TopMain --help ========================= 91s
1 tasks failed
xiangshan.resources os.SubprocessException: Result of git…: 1

    os.proc.call(ProcessOps.scala:230)
    build_.package_$xiangshan$.$anonfun$resources$3(build.sc:351)
make: *** [help] Error 1
[uno@myServer XiangShan]$ make jar
mill -i xiangshan.assembly
[677/721] xiangshan.resources
[677] You need to run this command from the toplevel of the working tree.
[674/721] openNCB.compile
[675/721] utility.compile
[674] [info] compiling 89 Scala sources to /home/xs/xs-env/XiangShan/out/openNCB/compile.dest/classes ...
[675] [info] compiling 58 Scala sources to /home/xs/xs-env/XiangShan/out/utility/compile.dest/classes ...
[675] [warn] /home/xs/xs-env/XiangShan/utility/src/main/scala/utility/PerfCounterUtils.scala:50:26: a pure expression does nothing in statement position
[675] [warn]       case regular(_) => true
[675] [warn]                          ^
[675] [warn] /home/xs/xs-env/XiangShan/utility/src/main/scala/utility/TLUtils/TLLogger.scala:91:5: match may not be exhaustive.
[675] [warn] It would fail on the following input: TLBundleE()
[675] [warn]     chn match {
[675] [warn]     ^
[675] [warn] 10 deprecations (since 2.13.0); re-run with -deprecation for details
[675] [warn] three warnings found
[675] [info] done compiling
[674] [info] done compiling
[721/721] ============================== xiangshan.assembly ============================== 24s
1 tasks failed
xiangshan.resources os.SubprocessException: Result of git…: 1

    os.proc.call(ProcessOps.scala:230)
    build_.package_$xiangshan$.$anonfun$resources$3(build.sc:351)
make: *** [jar] Error 1
```

cc0h, 2025-06-16:
[uno@myServer XiangShan]$ git submodule
 458f94be4c46b62e3c3be034bb18781d2271c129 ChiselAIA (native_axi-42-g458f94b)
 9c1d363b1899f4314b5a2b6be00fff8e109df06b coupledL2 (v0.9-269-g9c1d363)
 75aae957909975a31605d9f50e65ec26b7ee2a97 difftest (heads/master)
 e1bd4695ca7beb36a5ce7357e9527ad9e95b9ec1 fudian (heads/pipeline)
 7c9d4cbb2a6d315c701e81c2acf0a5ab9d9af5e4 huancun (heads/master)
 e41f5393d30f169ee2f142c429843a0c59062ef6 openLLC (e41f539)
 eca506ac0752506269c1540ebe779387df8e8d1e ready-to-run (heads/master)
 bcf2051a45fcd6c4b84a62a679888c2dae75cce4 rocket-chip (heads/master)
 8ef84f12393af527dfd1dc073549fa336332eff1 utility (heads/master)
 cadd3c2f43096e253d61296b33ac697be8354e29 yunsuan (remotes/origin/fix_pipeline-11-gcadd3c2)

[uno@myServer difftest]$ git log --oneline | head -1
75aae95 fix(HardwareError): use ExecutionGuide trigger hardware error exception (#625)

Tang-Haojin, 2025-06-16:
> > We need some more information. Could you provide the result of `git submodule` under `XiangShan` directory and the result of `git log --oneline | head -1` under `XiangShan/difftest` directory?
> 
> 你们复现了这个error吗？或者有没有什么临时的解决方案？我目前只需要编译出能用来给vcs仿真的verilog代码就好了。
> Have you reproduced this error? Or is there any temporary solution? For now, I just need to compile the Verilog code that can be used for VCS simulation.

You can try reverting https://github.com/OpenXiangShan/XiangShan/pull/3941.

Besides, this may be related to the version of git. Default git version of centos 7.9 is pretty old. You can also try to upgrade it yourself by compiling from source.