OpenXiangShan/XiangShan #4879: 关于制作workload的问题

han-jianing, 2025-07-10:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

想请教一下，我根据官方文档教程制作出了workload，在NEMU下运行，platform name显示是freechips，rocketchip-unknown。但运行ready-to-run下的linux.bin，platform name显示是xiangshan，xiangshan-kunminghu。这个原因是什么呢？如何修改呢？谢谢！

### What did you do before

根据官方文档教程制作出了workload，在NEMU下运行

### Environment

- NEMU commit id: 700011b8

cebarobot, 2025-07-10:
请检查设备树文件（*.dts 文件）中的配置。

Please check config in Device Tree file (*.dts file).

closed 2025-07-16 by han-jianing

han-jianing, 2025-07-16:
还想请教一个问题，我在NEMU上面运行文档教程自行生成的工作负载，没有问题，可以正常打印；但在香山核运行，并使用ready-to-run（commit id为：ed8c30e）进行比对，则会报错，
./build/emu -i $OPENSBI_HOME/build/platform/generic/firmware/fw_payload.bin  --diff ready-to-run/riscv64-nemu-interpreter-so 2> perf.log
报错为：
system_opcode_insn:Failed to access CSR 0x30d from M-mode
sbi_trap_error: hart0: trap0: illegal instruction handler failed (error -1)
通过上述方法运行linux.bin或coremark-2-iteration.bin均可以比对成功。

NewPaulWalker, 2025-07-16:
> 还想请教一个问题，我在NEMU上面运行文档教程自行生成的工作负载，没有问题，可以正常打印；但在香山核运行，并使用ready-to-run（commit id为：ed8c30e）下的仿真程序进行比对，则会报错， ./build/emu -i $OPENSBI_HOME/build/platform/generic/firmware/fw_payload.bin --diff ready-to-run/riscv64-nemu-interpreter-so 2> perf.log 报错为： system_opcode_insn:Failed to access CSR 0x30d from M-mode sbi_trap_error: hart0: trap0: illegal instruction handler failed (error -1) 通过上述方法运行linux.bin或coremark-2-iteration.bin均可以比对成功。

This is because the version of ready-to-run you are using is outdated. You can try updating it using Git. Additionally, for the cause of this issue, you can refer to https://github.com/OpenXiangShan/XiangShan/discussions/4848 and https://github.com/OpenXiangShan/NEMU/issues/885.

han-jianing, 2025-07-16:
收到，我会尝试一下，非常感谢！

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2025年07月16日 12:36 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>、State ***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] 关于制作workload的问题 (Issue #4879) |
NewPaulWalker left a comment (OpenXiangShan/XiangShan#4879)

还想请教一个问题，我在NEMU上面运行文档教程自行生成的工作负载，没有问题，可以正常打印；但在香山核运行，并使用ready-to-run（commit id为：ed8c30e）下的仿真程序进行比对，则会报错， ./build/emu -i $OPENSBI_HOME/build/platform/generic/firmware/fw_payload.bin --diff ready-to-run/riscv64-nemu-interpreter-so 2> perf.log 报错为： system_opcode_insn:Failed to access CSR 0x30d from M-mode sbi_trap_error: hart0: trap0: illegal instruction handler failed (error -1) 通过上述方法运行linux.bin或coremark-2-iteration.bin均可以比对成功。

This is because the version of ready-to-run you are using is outdated. You can try updating it using Git. Additionally, for the cause of this issue, you can refer to #4848 and OpenXiangShan/NEMU#885.