OpenXiangShan/XiangShan #4978: 运行linux5.10.3内核，出现错误

han-jianing, 2025-08-27:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

运行linux5.10.3内核，出现错误
error: hwirq 0xffff is too large for :cpus:cpu@0:interrupt-controller
运行linux6.10.3内核时不会有这个问题

### What did you do before

按照教程生成工作负载，使用linux6.10.3可以正常运行，但换成linux5.10.3后出现问题

### Environment

- XiangShan commit id:16ae9ddcd
- NEMU commit id:bbeddeac

### Additional context

Welcome to riscv64-NEMU!
For help, type "help"

OpenSBI v1.6-125-gc2671bb
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name               : freechips,rocketchip-unknown
Platform Features           : medeleg
Platform HART Count         : 1
Platform IPI Device         : aclint-mswi
Platform Timer Device       : aclint-mtimer @ 1000000Hz
Platform Console Device     : xlnx-uartlite
Platform HSM Device         : ---
Platform PMU Device         : ---
Platform Reboot Device      : ---
Platform Shutdown Device    : ---
Platform Suspend Device     : ---
Platform CPPC Device        : ---
Firmware Base               : 0x80000000
Firmware Size               : 317 KB
Firmware RW Offset          : 0x40000
Firmware RW Size            : 61 KB
Firmware Heap Offset        : 0x46000
Firmware Heap Size          : 37 KB (total), 2 KB (reserved), 12 KB (used), 23 KB (free)
Firmware Scratch Size       : 4096 B (total), 408 B (used), 3688 B (free)
Runtime SBI Version         : 3.0
Standard SBI Extensions     : ipi,pmu,sse,hsm,rfnc,fwft,time,base,legacy,dbcn,dbtr
Experimental SBI Extensions : none

Domain0 Name                : root
Domain0 Boot HART           : 0
Domain0 HARTs               : 0*
Domain0 Region00            : 0x0000000038000000-0x000000003800ffff M: (I,R,W) S/U: ()
Domain0 Region01            : 0x0000000080040000-0x000000008004ffff M: (R,W) S/U: ()
Domain0 Region02            : 0x0000000080000000-0x000000008003ffff M: (R,X) S/U: ()
Domain0 Region03            : 0x000000003c000000-0x000000003fffffff M: (I,R,W) S/U: (R,W)
Domain0 Region04            : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R,W,X)
Domain0 Next Address        : 0x0000000080200000
Domain0 Next Arg1           : 0x0000000082200000
[... 94 lines cut ...]
[    0.019869]  t1 : ffffffffffffffff t2 : 0000000000000400 s0 : ffffffe07feab990
[    0.020186]  s1 : ffffffe07fe0e200 a0 : 0000000000000049 a1 : ffffffe000848b58
[    0.020499]  a2 : 0000000000000000 a3 : 0000000000000000 a4 : 0000000000000000
[    0.020812]  a5 : 0000000000000000 a6 : 0000000000000000 a7 : 0000000000000001
[    0.021128]  s2 : 00000000ffffffff s3 : 00000000ffffffff s4 : 000000000000003f
[    0.021482]  s5 : ffffffe1f8df79d8 s6 : ffffffe00041ecf8 s7 : 0000000000000000
[    0.021797]  s8 : 0000000000000000 s9 : 0000000000000002 s10: ffffffe07fe19700
[    0.022112]  s11: 0000000000000000 t3 : ffffffe00040b2d0 t4 : 0000000000000402
[    0.022425]  t5 : ffffffe000848b90 t6 : ffffffe07feab698
[    0.022669] status: 8000000200000720 badaddr: ffffffe000249306 cause: 0000000000000003
[    0.023003] ---[ end trace a8a98e4e22f295a0 ]---
[    0.025406] clocksource: Switched to clocksource riscv_clocksource
[    0.037238] workingset: timestamp_bits=62 max_order=21 bucket_order=0
[    0.038808] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.041366] Freeing unused kernel memory: 144K
[    0.047361] Run /init as init process
Hello, RISC-V World!
[/home/hjn/xs-env/NEMU/src/isa/riscv64/include/../instr/special.h:38,execute] nemu_trap case 0
[src/cpu/cpu-exec.c:935,cpu_exec] nemu: HIT GOOD TRAP at pc = 0x0000000000010120
[src/cpu/cpu-exec.c:941,cpu_exec] trap code:0
[src/cpu/cpu-exec.c:155,monitor_statistic] host time spent = 1,058,147 us
[src/cpu/cpu-exec.c:157,monitor_statistic] total guest instructions = 147,627,200
[src/cpu/cpu-exec.c:158,monitor_statistic] vst count = 0, vst unit count = 0, vst unit optimized count = 0
[src/cpu/cpu-exec.c:161,monitor_statistic] simulation frequency = 139,514,831 instr/s
[src/utils/state.c:30,is_exit_status_bad] NEMU exit with good state: 2, halt ret: 0

cebarobot, 2025-08-27:
It looks like there may be an issue with your device tree configuration. Please check your device tree to see if any node has an interrupts property set too large.

Also, this is a Linux-related issue, not something directly related to XiangShan hardware. We cannot guarantee that we can accurately answer or solve this problem.

By the way, why are you using linux 5.10.3?

han-jianing, 2025-08-27:
好的 谢谢 博士大哥需要我使用基于5.10的

closed 2025-08-27 by han-jianing