OpenXiangShan/XiangShan #5024: Regarding support for the zicbom extension

BrainLyh, 2025-09-15:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

commit: 16ae9ddcda54fc9a2fddffad73174cf793ac7814

Has Xiangshan actually implemented the instructions defined in zicbom (e.g., cbo.flush) yet? Although Chapter 3 of the manual states that the extension is supported, an illegal-instruction exception is raised when I try to execute it.

```
==============  In the last commit group  ==============
the first commit instr pc of DUT is 0x000000008000020e
the first commit instr pc of REF is 0x000000008000020e

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 00800002e2 cmtcnt 2
commit group [01]: pc 0080000e0c cmtcnt 8
commit group [02]: pc 0080000f52 cmtcnt 11
commit group [03]: pc 0080000f6c cmtcnt 8
commit group [04]: pc 0080000f7c cmtcnt 1
commit group [05]: pc 0080001494 cmtcnt 1
commit group [06]: pc 0080001496 cmtcnt 1
commit group [07]: pc 0080001498 cmtcnt 1
commit group [08]: pc 008000020e cmtcnt 1 <--
commit group [09]: pc 00800002f2 cmtcnt 1
commit group [10]: pc 00800002f4 cmtcnt 3
commit group [11]: pc 00800002fc cmtcnt 1
commit group [12]: pc 0080000300 cmtcnt 3
commit group [13]: pc 00800002d6 cmtcnt 1
commit group [14]: pc 00800002da cmtcnt 1
commit group [15]: pc 00800002dc cmtcnt 2

============== Commit Instr Trace ==============
[00] commit pc 0000000080000f6c inst 0a013403 wen 1 dst 08 data 0000000000000000 idx 020 (05) ld      s0, 160(sp)
[01] commit pc 0000000080000f6e inst 09813483 wen 1 dst 09 data 0000000000000000 idx 021 (06) ld      s1, 152(sp)
[02] commit pc 0000000080000f70 inst 09013903 wen 1 dst 18 data 0000000080009f80 idx 022 (07) ld      s2, 144(sp)
[03] commit pc 0000000080000f72 inst 08813983 wen 1 dst 19 data 8000000a00006000 idx 023 (08) ld      s3, 136(sp)
[04] commit pc 0000000080000f74 inst 07013b03 wen 1 dst 22 data 0000000000000000 idx 024 (09) ld      s6, 112(sp)
[05] commit pc 0000000080000f76 inst 05013d03 wen 1 dst 26 data 0000000000000000 idx 025 (0a) ld      s10, 80(sp)
[06] commit pc 0000000080000f78 inst 04813d83 wen 1 dst 27 data 0000000000000000 idx 026 (0b) ld      s11, 72(sp)
[07] commit pc 0000000080000f7a inst 0b010113 wen 1 dst 02 data 0000000080009ee0 idx 027 addi    sp, sp, 176
[08] commit pc 0000000080000f7c inst 00008067 wen 0 dst 00 data 0000000000000000 idx 028 ret
[09] commit pc 0000000080001494 inst 01813083 wen 1 dst 01 data 000000008000020e idx 029 (0c) ld      ra, 24(sp)
[10] commit pc 0000000080001496 inst 06010113 wen 1 dst 02 data 0000000080009f40 idx 02a addi    sp, sp, 96
[11] commit pc 0000000080001498 inst 00008067 wen 0 dst 00 data 0000000000000000 idx 02b ret
[12] commit pc 000000008000020e inst 00090513 wen 1 dst 10 data 0000000080009f80 idx 02c mv      a0, s2
[13] exception pc 0000000080000210 inst 0024200f cause 0000000000000002 cbo.flush (s0) <--
[14] commit pc 00000000800002e2 inst 00a78223 wen 0 dst 04 data 0000000000000000 idx 00e (01) (S) sb      a0, 4(a5)
[15] commit pc 00000000800002e6 inst 00008067 wen 0 dst 00 data 0000000000000000 idx 00f ret
[16] commit pc 0000000080000e0c inst 001d8d93 wen 1 dst 27 data 000000000000001e idx 010 addi    s11, s11, 1
[17] commit pc 0000000080000e0e inst 000d4503 wen 1 dst 10 data 0000000000000000 idx 011 (16) lbu     a0, 0(s10)
[18] commit pc 0000000080000e12 inst 12050b63 wen 0 dst 22 data 0000000000000000 idx 012 beqz    a0, pc + 310
[19] commit pc 0000000080000f48 inst 08013a03 wen 1 dst 20 data 0000000000000000 idx 013 (17) ld      s4, 128(sp)
[20] commit pc 0000000080000f4a inst 07813a83 wen 1 dst 21 data 0000000000000000 idx 014 (00) ld      s5, 120(sp)
[21] commit pc 0000000080000f4c inst 06813b83 wen 1 dst 23 data 0000000000000000 idx 015 (01) ld      s7, 104(sp)
[22] commit pc 0000000080000f4e inst 06013c03 wen 1 dst 24 data 0000000000000000 idx 016 (02) ld      s8, 96(sp)
[23] commit pc 0000000080000f50 inst 05813c83 wen 1 dst 25 data 0000000000000000 idx 017 (03) ld      s9, 88(sp)
[24] commit pc 0000000080000f52 inst 000d841b wen 1 dst 08 data 000000000000001e idx 018 sext.w  s0, s11
[25] commit pc 0000000080000f56 inst 012de463 wen 0 dst 08 data 0000000000000000 idx 019 bltu    s11, s2, pc + 8
[26] commit pc 0000000080000f5e inst 00090693 wen 1 dst 13 data ffffffffffffffff idx 01a mv      a3, s2
[27] commit pc 0000000080000f66 inst 000480e7 wen 1 dst 01 data 0000000080000f68 idx 01b jalr    s1
[28] commit pc 0000000080000db6 inst 00051263 wen 0 dst 04 data 0000000000000000 idx 01c bnez    a0, pc + 4
[29] commit pc 0000000080000db8 inst 00008067 wen 0 dst 00 data 0000000000000000 idx 01d ret
[30] commit pc 0000000080000f68 inst 0a813083 wen 1 dst 01 data 0000000080001494 idx 01e (04) ld      ra, 168(sp)
[31] commit pc 0000000080000f6a inst 00040513 wen 1 dst 10 data 000000000000001e idx 01f mv      a0, s0
[... 60 lines cut ...]
v10: 0x665a5e77f7f816fc_08837473f5ec9813  v11: 0x6783c6676207d33b_eb648419dd9f3057  
v12: 0x380590666aaca5d3_200d0a40cc1be400  v13: 0x3083c53067ed04e2_0843a3d622fb91d5  
v14: 0x562c8062a4a41d11_34eea7ef3afe6ecb  v15: 0xdb8490534149899e_41d973888f12cf0c  
v16: 0x92b34d3f17c8bd1b_8e6f4f1aa531bbf0  v17: 0xcdb940e0edf6d6e1_bade34a3e4e96fcf  
v18: 0xb5fad14463a8d867_009c5b533053ca68  v19: 0x07f9d9f429f301b9_db4b74928c77128b  
v20: 0xc92743f33dc9ed80_3c98b3968c8fe974  v21: 0x8ce726c5a493b0cc_7cc916ab3cec1d52  
v22: 0xf78b5de303e8042a_d75921d9ae15c84a  v23: 0x44cf798ea4732c04_ecd22218e5204a46  
v24: 0xc4cee626386eaa04_fbe95e1bb7ce8fda  v25: 0x2967867442c5d4d0_0b86c83181306911  
v26: 0xaf7e97c01ffcfa97_e6438b3bb891e7e2  v27: 0x8999595104c2a808_b74e7f31c1ca6534  
v28: 0xb361d823641726f4_3e98b176491a17b9  v29: 0x419a27662037f12e_921eee6fa7515015  
v30: 0x017227d63f358fea_d8907fb0d31cc2ba  v31: 0x5a918845dcb0eabd_1b0041f052f8ef70  
  vtype: 0x8000000000000000 vstart: 0x0000000000000000  vxsat: 0x0000000000000000
   vxrm: 0x0000000000000003     vl: 0x0000000000000000   vcsr: 0x0000000000000006
---------------- Triggers ----------------
 tselect: 0x0000000000000000
 0: tdata1: 0xf000000000000000 tdata2: 0x0000000000000000
 1: tdata1: 0xf000000000000000 tdata2: 0x0000000000000000
 2: tdata1: 0xf000000000000000 tdata2: 0x0000000000000000
 3: tdata1: 0xf000000000000000 tdata2: 0x0000000000000000
 4: tdata1: 0x0000000000000000 tdata2: 0x0000000000000000
privilegeMode: 3
  mtval different at pc = 0x008000020e, right= 0x0000000000000000, wrong = 0x000000000024200f
 mcause different at pc = 0x008000020e, right= 0x0000000000000007, wrong = 0x0000000000000002
Core 0: [31mABORT at pc = 0x800002f4
```

ngc7331, 2025-09-15:
AFAIK zicbom is supported. Please refer to riscv spec Chapter 19.5. Table 23. Generic Format for xenvcfg CSRs:
> Cache Block Clean and Flush instruction Enable
> Enables the execution of the cache block clean instruction, CBO.CLEAN, and the cache block flush instruction, CBO.FLUSH, in a lower privilege mode: 
> 0: The instruction raises an illegal-instruction or virtual-instruction exception 
> 1: The instruction is executed

Have you configured `xenvcfg` CSR properly before executing `cbo.flush`?

BrainLyh, 2025-09-17:
@ngc7331 I have already set the value of menvcfg through code. ` asm volatile("csrw menvcfg, %0" :: "r"(0x40));`, So this part should be fine.

@Yan-Muzi  Regarding the compilation command, do you mean the one used for the test code or the one for NEMU itself?
For the test code I added the following line to riscv64.mk `MARCH ?= rv64gc_zicbom` and then built with `make ARCH=riscv64-xs`
I did not re-compile NEMU.

If there’s any other information you still need, please let me know right away.

Yan-Muzi, 2025-09-17:
It's the command you used to build XiangShan. `DefaultConfig` does not support zicbom while `KunminghuV2Config` does.

Ma-YX, 2025-09-25:
There seems to be two possible problems with your test program:
1. `rdcycle` itself is not a strong serialization barrier; on an OOO core, surrounding instructions may be reordered and execute around it. You can use `fence` to ensure sequence.
2. Sequential access in test has strong spatial locality, making the test results prone to prefetcher interference.

Here is optimized test(mostly generated by LLM, question if any problem):
```c
#include <stdint.h>
#include <stdio.h>
#include <klib.h>

#ifndef __riscv_zicbom
#error "This benchmark requires Zicbom. Compile with -march=rv64gc_zicbom (or rv32gc_zicbom)."
#endif

#ifndef CACHE_LINE_SIZE
#define CACHE_LINE_SIZE 64
#endif

// ------------------------- 基本内联工具 -------------------------

static inline unsigned long rdcycle(void) {
  unsigned long x;
  asm volatile ("csrr %0, cycle" : "=r"(x));
  return x;
}

static inline unsigned long csr_read_menvcfg(void) {
  unsigned long val;
  asm volatile ("csrr %0, menvcfg" : "=r"(val));
  return val;
}
static inline void csr_set_menvcfg(unsigned long val) {
  asm volatile ("csrw menvcfg, %0" :: "r"(val));
}

// Zicbom：写回并失效指定 cache line
static inline void cbo_flush(void *addr) {
  asm volatile ("cbo.flush (%0)" :: "r"(addr) : "memory");
  asm volatile ("fence rw,rw" ::: "memory");  // 约束完成
}

// ------------------------- 强序列化测量原语 -------------------------

// 被测：对 *p 的一次 load（带数据依赖），再对 *sink 的 store 依赖该值；
// fence rw,rw 迫使写完成，从而把 t1 钉在 load 之后。
static inline unsigned long load_cycles_dep_store(const volatile uint8_t *p,
                                                  volatile uint8_t *sink) {
  unsigned long t0, t1;
  uint8_t v;
  asm volatile(
      "fence rw,rw\n"
      "csrr  %0, cycle\n"      // t0
      "lbu   %2, 0(%3)\n"      // v = *p   （被测 load）
      "addi  %2, %2, 1\n"      // 轻量数据依赖
      "sb    %2, 0(%4)\n"      // 依赖 v 的 store
      "fence rw,rw\n"          // 等待前述读写完成
      "csrr  %1, cycle\n"      // t1
      : "=&r"(t0), "=&r"(t1), "=&r"(v)
      : "r"(p), "r"(sink)
      : "memory");
  return t1 - t0;
}

// 框架基线（不含被测 load），用于扣除常数项
static inline unsigned long baseline_cycles(volatile uint8_t *sink) {
  unsigned long t0, t1;
  uint8_t v;
  asm volatile(
      "fence rw,rw\n"
      "csrr  %0, cycle\n"
[... 102 lines cut ...]
menvcfg original=0x80000000000000f0, now=0x80000000000000f0  (bit6 CBCFE may be WARL)
Baseline cycles: 46

Index | Data |  Warm |  Hot  |  Diff(warm-hot)
------+------|-------|-------|----------------
   0  |  A   |   121 |     1 |            120
   1  |  B   |   119 |     1 |            118
   2  |  C   |   120 |     1 |            119
   3  |  D   |   119 |     1 |            118
   4  |  E   |   120 |     1 |            119
   5  |  F   |   120 |     1 |            119
   6  |  G   |   120 |     1 |            119
   7  |  H   |   120 |     1 |            119
   8  |  I   |   120 |     1 |            119
   9  |  J   |   120 |     1 |            119
  10  |  K   |   120 |     1 |            119
  11  |  L   |   119 |     1 |            118
  12  |  M   |   120 |     1 |            119
  13  |  N   |   120 |     1 |            119
  14  |  O   |   120 |     1 |            119
  15  |  P   |   120 |     1 |            119

Avg    Warm=119  Hot=1  Diff(warm-hot)=118
Elements with warm > hot: 16/16 (100.0%)
```